FASTSHIFT/Arduino-For-Keil #3: Adafruit_SSD1306 例子报错

konosubakonoakua, 2019-03-26:
先谢谢大佬的开源，我尝试编译了一下oled的库，有点小问题
Compiler：ac5
IDE：KEIL v5.27pre
错误信息：
`..\Libraries\Adafruit_SSD1306\Adafruit_SSD1306.cpp(184): error:  #513: a value of type "volatile std::uint32_t *" cannot be assigned to an entity of type "PortReg *"`
解决方法：
添加强制类型转换 `(PortReg *)`
```
        csport      = (PortReg*)portOutputRegister(digitalPinToPort(cs));
        cspinmask   = digitalPinToBitMask(cs);
        dcport      = (PortReg*)portOutputRegister(digitalPinToPort(dc));
        dcpinmask   = digitalPinToBitMask(dc);
```

FASTSHIFT, 2019-03-26:
您使用的是STM32F0系列吗？
这个 portOutputRegister(digitalPinToPort(pin))函数 返回的是GPIO的ODR寄存器地址。
STM32F0系列的单片机的GPIO的ODR寄存器是16位的，而其他STM32系列是32位的，所以才有这个报错。
建议解决办法：
在Adafruit_SSD1306.h文件里的第45行:

`#elif defined(__STM32__)
 typedef volatile uint32_t PortReg;
typedef uint16_t PortMask;
#define HAVE_PORTREG
#else`

修改为：
`#elif defined(__STM32__)
    #if defined(__STM32F0__)
        typedef volatile uint16_t PortReg;
    #else
        typedef volatile uint32_t PortReg;
    #endif
        typedef uint16_t PortMask;
    #define HAVE_PORTREG
#else`

不建议使用强制转换

konosubakonoakua, 2019-03-26:
我用的f1🤣, 不过香您这样改要更好

FASTSHIFT, 2019-03-26:
@konosubakonoakua 
感谢提交问题，Adafruit_SSD1306库已重新上传，您可以再编译一下试试

GavinOUYANG, 2019-06-05:
> @konosubakonoakua
> 感谢提交问题，Adafruit_SSD1306库已重新上传，您可以再编译一下试试

请问为什么编译成功但是就没有图案显示

FASTSHIFT, 2019-06-06:
@GavinOUYANG 尝试多初始化几次

GavinOUYANG, 2019-06-07:
> @GavinOUYANG 尝试多初始化几次

不知道大佬有没有添加号显示器库的历程，试过u8g2也是不行，谢谢了

GavinOUYANG, 2019-06-07:
> @GavinOUYANG 尝试多初始化几次

就简单显示一个英文就行了，搞了两三天还是不行，谢谢了

FASTSHIFT, 2019-06-08:
@GavinOUYANG 库自带的例程就可以用

FASTSHIFT, 2019-06-08:
@GavinOUYANG 您用的是I2C屏吗，可以尝试降低I2C速度，在Wire.h文件里有一个#define FULL_SPEED_I2C，把它注释掉。还有检查一下屏幕的I2C地址，是否和库设置的地址匹配。

GavinOUYANG, 2019-06-10:
> @GavinOUYANG 您用的是I2C屏吗，可以尝试降低I2C速度，在Wire.h文件里有一个#define FULL_SPEED_I2C，把它注释掉。还有检查一下屏幕的I2C地址，是否和库设置的地址匹配。

好的谢谢

closed 2021-10-13 by FASTSHIFT