FASTSHIFT/Arduino-For-Keil #4: Servo库频率错误

EJDRONE, 2019-05-08:
您好
首先感谢大神做出的辛勤劳动，这样的项目意义重大，昨天下载下来后捣鼓一下就成功了

首先报告我用的是STM32F407VET6，代码很简单，就是用的Servo库例程
我在实验Servo库时，按顺序实验了TIM1~TIM5 挂接的管脚，发现除了使用TIM1 `时的频率与占空比是正确的（50HZ）
**见图（TIM1,PA10）：**
[image]

使用TIM 2，3挂接的管脚都出现了占空比正确，但频率不对的问题（25HZ）
**见图（TIM2，PA0）**
[image]

FASTSHIFT, 2019-05-09:
感谢您提交的BUG，Servo库的错误已修复。
[image]

频率错误的原因是因为STM32的TIM1的外设时钟是挂在APB2上的，而TIM2~TIM5的时钟是挂在APB1上的，APB1的时钟频率只有APB2的一半。而Servo库默认的模式是APB2，所以使用TIM2~TIM5输出PWM的时候，输出的频率只有TIM1的一半。
[image]

EJDRONE, 2019-05-09:
十分感谢大神，经测试，已经OK！
我今天试了一下直接使用U8G2库，会报错，主要是两个方面的错误：
一个是yield()函数未定义，因为U8x8lib.cpp文件的u8x8_gpio_and_delay_arduino()函数会调用到yield()函数，这个yield()我查了一下，声明在标准arduino中的Arduino.h里，而您的Arduino.h是没有这个函数的，当然我没去弄清楚这个函数的作用，所以就把U8G2库里对yield()的调用注释掉了，错误消失。

第二个是一些函数中变量声明的顺序的错误，我调整U8G2库里的声明顺序后，错误消失。

剩下的一些编译警告我未作处理，但烧进STM32F407后，是可以正常显示的，包括使用SPI 1带两块屏，没看出问题，当然我对我所做的这些改动后果未知，按原则来说也不应该改库，只是给您一个报告，我将会继续使用您的项目做实验，会继续提出BUG

FASTSHIFT, 2019-05-09:
感谢您的支持@EJDRONE 
我也会继续优化提升兼容性

closed 2021-10-13 by FASTSHIFT