FASTSHIFT/Arduino-For-Keil #6: _Keilduino (STM32F4xx)打开Keil，报错。

Seninthefool, 2019-07-15:
错误如下：Device下各个组件为红色报错
Warning #440: Requested device 'STM32F407VE' is subsituted with variant 'STM32F407VETx' for target 'Project'
Error instantiating RTE components
Error #540: 'Keil::Device:StdPeriph Drivers:ADC:1.3.0' component is not available for target 'Project', pack 'Keil.STM32F4xx_DFP.1.0.8' is not selected
Error #540: 'Keil::Device:StdPeriph Drivers:DCMI:1.3.0' component is not available for target 'Project', pack 'Keil.STM32F4xx_DFP.1.0.8' is not selected
Error #540: 'Keil::Device:StdPeriph Drivers:DMA:1.3.0' component is not available for target 'Project', pack 'Keil.STM32F4xx_DFP.1.0.8' is not selected
Error #540: 'Keil::Device:StdPeriph Drivers:EXTI:1.3.0' component is not available for target 'Project', pack 'Keil.STM32F4xx_DFP.1.0.8' is not selected
Error #540: 'Keil::Device:StdPeriph Drivers:Flash:1.3.0' component is not available for target 'Project', pack 'Keil.STM32F4xx_DFP.1.0.8' is not selected
Error #540: 'Keil::Device:StdPeriph Drivers:Framework:1.3.1' component is not available for target 'Project', pack 'Keil.STM32F4xx_DFP.1.0.8' is not selected
Error #540: 'Keil::Device:StdPeriph Drivers:GPIO:1.3.0' component is not available for target 'Project', pack 'Keil.STM32F4xx_DFP.1.0.8' is not selected
Error #540: 'Keil::Device:StdPeriph Drivers:I2C:1.3.0' component is not available for target 'Project', pack 'Keil.STM32F4xx_DFP.1.0.8' is not selected
Error #540: 'Keil::Device:StdPeriph Drivers:RCC:1.3.0' component is not available for target 'Project', pack 'Keil.STM32F4xx_DFP.1.0.8' is not selected
Error #540: 'Keil::Device:StdPeriph Drivers:SPI:1.3.0' component is not available for target 'Project', pack 'Keil.STM32F4xx_DFP.1.0.8' is not selected
Error #540: 'Keil::Device:StdPeriph Drivers:SYSCFG:1.3.0' component is not available for target 'Project', pack 'Keil.STM32F4xx_DFP.1.0.8' is not selected
Error #540: 'Keil::Device:StdPeriph Drivers:TIM:1.3.0' component is not available for target 'Project', pack 'Keil.STM32F4xx_DFP.1.0.8' is not selected
Error #540: 'Keil::Device:StdPeriph Drivers:USART:1.3.0' component is not available for target 'Project', pack 'Keil.STM32F4xx_DFP.1.0.8' is not selected

FASTSHIFT, 2019-07-15:
标准库已经正确安装了吗？

Seninthefool, 2019-07-15:
之前安装的是Keil.STM32F4xx_DFP.2.13.0的pack包，刚刚替换了您发布的1.0.8的pack包，错误还存在。

Seninthefool, 2019-07-15:
组件不可用于目标项目

Seninthefool, 2019-07-15:
取消勾选组件使用即可，并且启动文件需要更新。更新后报如下错误
*** Using Compiler 'V5.06 update 6 (build 750)', folder: 'G:\ARM\ARMCC\Bin'
Build target 'Project'
compiling dtostrf.c...
compiling itoa.c...
compiling WMath.cpp...
compiling Print.cpp...
compiling WString.cpp...
compiling Stream.cpp...
..\Core\mcu_type.h(5): error:  #5: cannot open source input file "stm32f4xx_conf.h": No such file or directory
  #include "stm32f4xx_conf.h"
..\ArduinoAPI\Stream.cpp: 0 warnings, 1 error
compiling Tone.cpp...
..\Core\mcu_type.h(5): error:  #5: cannot open source input file "stm32f4xx_conf.h": No such file or directory
  #include "stm32f4xx_conf.h"
..\ArduinoAPI\Tone.cpp: 0 warnings, 1 error
compiling Wire.cpp...
..\Core\mcu_type.h(5): error:  #5: cannot open source input file "stm32f4xx_conf.h": No such file or directory
  #include "stm32f4xx_conf.h"
..\ArduinoAPI\Wire.cpp: 0 warnings, 1 error
compiling WireBase.cpp...
..\Core\mcu_type.h(5): error:  #5: cannot open source input file "stm32f4xx_conf.h": No such file or directory
  #include "stm32f4xx_conf.h"
..\ArduinoAPI\WireBase.cpp: 0 warnings, 1 error
compiling HardwareSerial.cpp...
..\Core\mcu_type.h(5): error:  #5: cannot open source input file "stm32f4xx_conf.h": No such file or directory
  #include "stm32f4xx_conf.h"
..\ArduinoAPI\HardwareSerial.cpp: 0 warnings, 1 error
compiling Arduino.c...
..\Core\mcu_type.h(5): error:  #5: cannot open source input file "stm32f4xx_conf.h": No such file or directory
  #include "stm32f4xx_conf.h"
..\ArduinoAPI\Arduino.c: 0 warnings, 1 error
compiling SPI.cpp...
..\Core\mcu_type.h(5): error:  #5: cannot open source input file "stm32f4xx_conf.h": No such file or directory
  #include "stm32f4xx_conf.h"
..\ArduinoAPI\SPI.cpp: 0 warnings, 1 error
compiling adc.c...
..\Core\mcu_type.h(5): error:  #5: cannot open source input file "stm32f4xx_conf.h": No such file or directory
  #include "stm32f4xx_conf.h"
..\Core\adc.c: 0 warnings, 1 error
assembling startup_stm32f407xx.s...
compiling delay.c...
..\CORE\mcu_type.h(5): error:  #5: cannot open source input file "stm32f4xx_conf.h": No such file or directory
  #include "stm32f4xx_conf.h"
..\CORE\delay.c: 0 warnings, 1 error
compiling gpio.c...
..\Core\mcu_type.h(5): error:  #5: cannot open source input file "stm32f4xx_conf.h": No such file or directory
  #include "stm32f4xx_conf.h"
..\Core\gpio.c: 0 warnings, 1 error
compiling timer.c...
..\Core\mcu_type.h(5): error:  #5: cannot open source input file "stm32f4xx_conf.h": No such file or directory
  #include "stm32f4xx_conf.h"
..\Core\timer.c: 0 warnings, 1 error
compiling pwm.c...
..\Core\mcu_type.h(5): error:  #5: cannot open source input file "stm32f4xx_conf.h": No such file or directory
  #include "stm32f4xx_conf.h"
..\Core\pwm.c: 0 warnings, 1 error
compiling main.cpp...
..\Core\mcu_type.h(5): error:  #5: cannot open source input file "stm32f4xx_conf.h": No such file or directory
  #include "stm32f4xx_conf.h"
main.cpp: 0 warnings, 1 error
compiling exti.c...
..\Core\mcu_type.h(5): error:  #5: cannot open source input file "stm32f4xx_conf.h": No such file or directory
  #include "stm32f4xx_conf.h"
..\Core\exti.c: 0 warnings, 1 error
compiling system_stm32f4xx.c...
"..\OBJ\Pro.axf" - 14 Error(s), 0 Warning(s).
Target not created.
Build Time Elapsed:  00:00:29

FASTSHIFT, 2019-07-15:
工程里应该有这个头文件，重新添加一下

Seninthefool, 2019-07-15:
目前没有F401Re的Device，请问呢能用STM32F401CC或STM32F401CB代替吗

Seninthefool, 2019-07-15:
添加完成后，又有新错误
*** Using Compiler 'V5.06 update 6 (build 750)', folder: 'G:\ARM\ARMCC\Bin'
Build target 'Project'
compiling WMath.cpp...
compiling dtostrf.c...
compiling itoa.c...
compiling Print.cpp...
compiling WString.cpp...
compiling SPI.cpp...
.\RTE\Device\STM32F401CB\stm32f4xx_conf.h(90): error:  #5: cannot open source input file "misc.h": No such file or directory
  #include "misc.h" /* High level functions for NVIC and SysTick (add-on to CMSIS functions) */
..\ArduinoAPI\SPI.cpp: 0 warnings, 1 error
compiling Arduino.c...
.\RTE\Device\STM32F401CB\stm32f4xx_conf.h(90): error:  #5: cannot open source input file "misc.h": No such file or directory
  #include "misc.h" /* High level functions for NVIC and SysTick (add-on to CMSIS functions) */
..\ArduinoAPI\Arduino.c: 0 warnings, 1 error
compiling WireBase.cpp...
.\RTE\Device\STM32F401CB\stm32f4xx_conf.h(90): error:  #5: cannot open source input file "misc.h": No such file or directory
  #include "misc.h" /* High level functions for NVIC and SysTick (add-on to CMSIS functions) */
..\ArduinoAPI\WireBase.cpp: 0 warnings, 1 error
compiling Stream.cpp...
.\RTE\Device\STM32F401CB\stm32f4xx_conf.h(90): error:  #5: cannot open source input file "misc.h": No such file or directory
  #include "misc.h" /* High level functions for NVIC and SysTick (add-on to CMSIS functions) */
..\ArduinoAPI\Stream.cpp: 0 warnings, 1 error
compiling HardwareSerial.cpp...
.\RTE\Device\STM32F401CB\stm32f4xx_conf.h(90): error:  #5: cannot open source input file "misc.h": No such file or directory
  #include "misc.h" /* High level functions for NVIC and SysTick (add-on to CMSIS functions) */
..\ArduinoAPI\HardwareSerial.cpp: 0 warnings, 1 error
compiling Tone.cpp...
.\RTE\Device\STM32F401CB\stm32f4xx_conf.h(90): error:  #5: cannot open source input file "misc.h": No such file or directory
  #include "misc.h" /* High level functions for NVIC and SysTick (add-on to CMSIS functions) */
..\ArduinoAPI\Tone.cpp: 0 warnings, 1 error
compiling Wire.cpp...
.\RTE\Device\STM32F401CB\stm32f4xx_conf.h(90): error:  #5: cannot open source input file "misc.h": No such file or directory
  #include "misc.h" /* High level functions for NVIC and SysTick (add-on to CMSIS functions) */
..\ArduinoAPI\Wire.cpp: 0 warnings, 1 error
compiling gpio.c...
.\RTE\Device\STM32F401CB\stm32f4xx_conf.h(90): error:  #5: cannot open source input file "misc.h": No such file or directory
  #include "misc.h" /* High level functions for NVIC and SysTick (add-on to CMSIS functions) */
..\Core\gpio.c: 0 warnings, 1 error
compiling adc.c...
.\RTE\Device\STM32F401CB\stm32f4xx_conf.h(90): error:  #5: cannot open source input file "misc.h": No such file or directory
  #include "misc.h" /* High level functions for NVIC and SysTick (add-on to CMSIS functions) */
..\Core\adc.c: 0 warnings, 1 error
compiling delay.c...
.\RTE\Device\STM32F401CB\stm32f4xx_conf.h(90): error:  #5: cannot open source input file "misc.h": No such file or directory
  #include "misc.h" /* High level functions for NVIC and SysTick (add-on to CMSIS functions) */
..\CORE\delay.c: 0 warnings, 1 error
compiling exti.c...
.\RTE\Device\STM32F401CB\stm32f4xx_conf.h(90): error:  #5: cannot open source input file "misc.h": No such file or directory
  #include "misc.h" /* High level functions for NVIC and SysTick (add-on to CMSIS functions) */
..\Core\exti.c: 0 warnings, 1 error
compiling timer.c...
.\RTE\Device\STM32F401CB\stm32f4xx_conf.h(90): error:  #5: cannot open source input file "misc.h": No such file or directory
  #include "misc.h" /* High level functions for NVIC and SysTick (add-on to CMSIS functions) */
..\Core\timer.c: 0 warnings, 1 error
compiling pwm.c...
.\RTE\Device\STM32F401CB\stm32f4xx_conf.h(90): error:  #5: cannot open source input file "misc.h": No such file or directory
  #include "misc.h" /* High level functions for NVIC and SysTick (add-on to CMSIS functions) */
..\Core\pwm.c: 0 warnings, 1 error
compiling main.cpp...
.\RTE\Device\STM32F401CB\stm32f4xx_conf.h(90): error:  #5: cannot open source input file "misc.h": No such file or directory
  #include "misc.h" /* High level functions for NVIC and SysTick (add-on to CMSIS functions) */
main.cpp: 0 warnings, 1 error
compiling system_stm32f4xx.c...
"..\OBJ\Pro.axf" - 14 Error(s), 0 Warning(s).
Target not created.
Build Time Elapsed:  00:00:23

Seninthefool, 2019-07-15:
缺少misc.h的头文件

FASTSHIFT, 2019-07-15:
[image]
查看工程配置是否正确

FASTSHIFT, 2019-07-15:
F401系列应该都差不多，只有RAM和FLASH容量的区别

Seninthefool, 2019-07-15:
想用QQ请教下，请问大牛QQ多少

FASTSHIFT, 2019-07-15:
在我的个人简介里有个邮箱地址，那个就是

closed 2021-10-13 by FASTSHIFT