FASTSHIFT/Arduino-For-Keil #8: 加载extEEPROM编译错误

whqsz, 2019-10-27:
在当前更新空模板上，项目只加载extEEPROM库后预编译提示11个错误，错误信息
compiling extEEPROM.cpp...
..\ArduinoAPI\WString.h(174): error:  #338: more than one instance of overloaded function "operator+"  has "C" linkage
      friend StringSumHelper & operator + (const StringSumHelper &lhs, const char *cstr);
..\ArduinoAPI\WString.h(175): error:  #338: more than one instance of overloaded function "operator+"  has "C" linkage
      friend StringSumHelper & operator + (const StringSumHelper &lhs, char c);
..\ArduinoAPI\WString.h(176): error:  #338: more than one instance of overloaded function "operator+"  has "C" linkage
      friend StringSumHelper & operator + (const StringSumHelper &lhs, unsigned char num);
..\ArduinoAPI\WString.h(177): error:  #338: more than one instance of overloaded function "operator+"  has "C" linkage
      friend StringSumHelper & operator + (const StringSumHelper &lhs, int num);
..\ArduinoAPI\WString.h(178): error:  #338: more than one instance of overloaded function "operator+"  has "C" linkage
      friend StringSumHelper & operator + (const StringSumHelper &lhs, unsigned int num);
..\ArduinoAPI\WString.h(179): error:  #338: more than one instance of overloaded function "operator+"  has "C" linkage
      friend StringSumHelper & operator + (const StringSumHelper &lhs, long num);
..\ArduinoAPI\WString.h(180): error:  #338: more than one instance of overloaded function "operator+"  has "C" linkage
      friend StringSumHelper & operator + (const StringSumHelper &lhs, unsigned long num);
..\ArduinoAPI\WString.h(181): error:  #338: more than one instance of overloaded function "operator+"  has "C" linkage
      friend StringSumHelper & operator + (const StringSumHelper &lhs, float num);
..\ArduinoAPI\WString.h(182): error:  #338: more than one instance of overloaded function "operator+"  has "C" linkage
      friend StringSumHelper & operator + (const StringSumHelper &lhs, double num);
..\ArduinoAPI\WString.h(183): error:  #338: more than one instance of overloaded function "operator+"  has "C" linkage
      friend StringSumHelper & operator + (const StringSumHelper &lhs, const __FlashStringHelper *rhs);
..\ArduinoAPI\Print.h(100): error:  #800: this declaration may not have extern "C" linkage
  template<class T> inline Print &operator << (Print &obj, T arg) {
..\Libraries\extEEPROM\extEEPROM.cpp: 0 warnings, 11 errors
compiling Tone.cpp...
compiling adc.c...
compiling delay.c...
assembling startup_stm32f10x_md.s...
compiling exti.c...
compiling gpio.c...
compiling pwm.c...
compiling timer.c...
compiling main.cpp...
compiling system_stm32f10x.c...
compiling misc.c...
compiling stm32f10x_adc.c...
compiling stm32f10x_dma.c...
compiling stm32f10x_exti.c...
compiling stm32f10x_flash.c...
compiling stm32f10x_gpio.c...
compiling stm32f10x_rcc.c...
compiling stm32f10x_spi.c...
compiling stm32f10x_usart.c...
compiling stm32f10x_tim.c...
"..\OBJ\Pro.axf" - 11 Error(s), 0 Warning(s).
Target not created.

在上个更新模板下加载extEEPROM过是正常编译的。

whqsz, 2019-10-27:
对比两个库唯一区别是外部调用的声明这个地方
//extern"C"{
#include "Arduino.h"
//}
#include "Wire.h"

FASTSHIFT, 2019-10-27:
@whqsz 新的extEEPROM库已上传，而且增加了示例

[image]
去除 extEEPROM.h 的 "extern"C"{}" 即可

whqsz, 2019-10-28:
感谢你的回复，问题已经解决。
目前对DigitalFilter.h调用上还是弄不明白，最好提交一份demo参考，让我们新手少走弯路。
谢谢！

FASTSHIFT, 2019-10-28:
好的，我稍后会更新例程

FASTSHIFT, 2019-10-28:
@whqsz 数字滤波器例程已更新

whqsz, 2019-11-11:
遇到一个问题，当单片机不外接8M晶振系统默认运行速度很慢，项目核心库中暂时没找到启用内部64M频率的函数，但我只找到这个函数来设置内部晶振。我希望能尽量使用本项目自身封装函数。感谢！
void SetSysClockTo64Mhz(void)  //启用内部晶振工作频率64M
{  
  __IO uint32_t StartUpCounter = 0, HSEStatus = 0;  
    RCC_DeInit();  
  /* SYSCLK, HCLK, PCLK2 and PCLK1 configuration ---------------------------*/      
  /* Enable HSI */      

  RCC->CR |= ((uint32_t)RCC_CR_HSION);  
  /* Wait till HSI is ready and if Time out is reached exit */  

  do  
  {  

    HSEStatus = RCC->CR & RCC_CR_HSIRDY;  
    StartUpCounter++;    
  } while((HSEStatus == 0) && (StartUpCounter != HSE_STARTUP_TIMEOUT));  
  if ((RCC->CR & RCC_CR_HSIRDY) != RESET)  
  { 
    HSEStatus = (uint32_t)0x01;  
  }  
  else  
  {  
    HSEStatus = (uint32_t)0x00;  
  }    
  if (HSEStatus == (uint32_t)0x01)  
  {  
    /* Enable Prefetch Buffer */  
    FLASH->ACR |= FLASH_ACR_PRFTBE;  
    /* Flash 2 wait state */  
    FLASH->ACR &= (uint32_t)((uint32_t)~FLASH_ACR_LATENCY);  
    FLASH->ACR |= (uint32_t)FLASH_ACR_LATENCY_2;      
    /* HCLK = SYSCLK */  
    RCC->CFGR |= (uint32_t)RCC_CFGR_HPRE_DIV1;  
    /* PCLK2 = HCLK */  
    RCC->CFGR |= (uint32_t)RCC_CFGR_PPRE2_DIV1;  
    /* PCLK1 = HCLK */  
    RCC->CFGR |= (uint32_t)RCC_CFGR_PPRE1_DIV2;  
    /*  PLL configuration: PLLCLK = HSI/2 * 16 = 64 MHz */  
    RCC->CFGR &= (uint32_t)((uint32_t)~(RCC_CFGR_PLLSRC | RCC_CFGR_PLLXTPRE |  
                                        RCC_CFGR_PLLMULL));  
    RCC->CFGR |= (uint32_t)(RCC_CFGR_PLLSRC_HSI_Div2| RCC_CFGR_PLLMULL16);  
    /* Enable PLL */  
    RCC->CR |= RCC_CR_PLLON;  
    /* Wait till PLL is ready */  
    while((RCC->CR & RCC_CR_PLLRDY) == 0)  
    {  
    }  
    /* Select PLL as system clock source */  
    RCC->CFGR &= (uint32_t)((uint32_t)~(RCC_CFGR_SW));  

    RCC->CFGR |= (uint32_t)RCC_CFGR_SW_PLL;      
    /* Wait till PLL is used as system clock source */  
    while ((RCC->CFGR & (uint32_t)RCC_CFGR_SWS) != (uint32_t)0x08)  
    {  
    }  
  }  
  else  
  { /* If HSE fails to start-up, the application will have wrong clock  
        configuration. User can add here some code to deal with this error */  

  }  

}

FASTSHIFT, 2019-11-11:
使用内部时钟源属于特殊需求，通用性不强，所以我没有在项目里封装，用标准库就可以了

FASTSHIFT, 2019-11-11:
在配置完时钟源后，调用Delay_Init()，让Systick自适应

whqsz, 2019-11-20:
这两次更新 timer.c
只要在main中加上定时器中断 程序运行卡在setup过不去 进程追踪停止while(System_ms < Stop_TimePoint);这行
/**
  * @brief  毫秒级延时
  * @param  ms: 要延时的毫秒数
  * @retval 无
  */
void delay_ms(uint32_t ms)
{
    uint32_t Stop_TimePoint = System_ms + ms;
    while(System_ms < Stop_TimePoint);   **停在这里**  
}
---------------------------------------------------------------------------------------------
项目只用到Wire库

#include "FileGroup.h"
...
...
int main(void)
{
	SetSysClockTo64Mhz();
	NVIC_PriorityGroupConfig(NVIC_PriorityGroup_2);
	GPIO_JTAG_Disable();
	//SysClock_Init(F_CPU_64MHz);
	Delay_Init();
	//ADCx_Init(ADC1);0
	TimerSet(TIM2, 1000, Onems); //1毫秒定时器   **这里**
	TIM_Cmd(TIM2, ENABLE);      **这里**                  
	
	setup();
	for (;;)loop();
}

FASTSHIFT, 2019-11-20:
@whqsz 加我Q吧

FASTSHIFT, 2019-11-20:
是不是在中断里调用delay了