FASTSHIFT/Arduino-For-Keil #9: HAL end CUBEMX

promavto, 2019-12-08:
Hi! Planning to use the HAL library and CUBEMX?
Alexander.
promavto@ntmp.ru

FASTSHIFT, 2019-12-08:
Yes, but not now.
Although it is really convenient to configure the project with CubeMX, I don't like some of the design of CubeMX:
1.CubeMX always adds stm32xxx_it.c to the project, causing conflicts with my library.
2. The single-byte sending speed of the SPI library of HAL is slow, making the STM32F7 run as slow as STM32F0. I don't want to use this to repackage the SPI library of Arduino.
3. The HAL library compiles slowly on Keil (AC5 compiler).

But in the future, I will still use HAL. Using CubeMX can easily initialize some complex peripherals and easily check the usage of peripherals. ST's new microcontrollers only provide HAL / LL libraries.
I have tried to use the LL library before(https://github.com/FASTSHIFT/Arduino-For-STM32Cube) , but there is too little reference material, so this project has been temporarily suspended, but it will be re-enabled in the future.

promavto, 2019-12-09:
Thanks! I understood your arguments. Very interesting project. Planning to use the emWin library

FASTSHIFT, 2019-12-09:
You can take a look at the LittlevGL library (https://github.com/littlevgl/lvgl), I think it is a very good open source GUI framework.
My other project (https://github.com/FASTSHIFT/XC-OS) is based on this GUI library.

promavto, 2019-12-09:
Ooh!!!  Very interesting, I have been looking for such information for a long time. I will carefully study your work.