FASTSHIFT/Arduino-For-Keil #12: millis()多线程运行时间不对

kehale, 2020-04-02:
我用millis()多线程运行时间，发现运行millis()时间不对，慢的很多。 
const int ledPin =  PC13; // the LED pin number connected
int ledState = LOW;             // used to set the LED state
unsigned long previousMillis = 0;  //will store last time LED was blinked
const long period = 300; 
 unsigned long currentMillis = millis(); // store the current time
 
pinMode(PC13, OUTPUT); 

 if (currentMillis - previousMillis >= period) { // check if 1000ms passed
   previousMillis = currentMillis;   // save the last time you blinked the LED
   if (ledState == LOW) { // if the LED is off turn it on and vice-versa
                                     ledState = HIGH;
                                     } else {
                                    ledState = LOW;
                                  }
   digitalWrite(ledPin, ledState);//set LED with ledState to blink again
 }	

上面是我的代码，其中const long period = 300; 这个原来是1000ms，我发现PC13 的led灯闪烁间隔有7秒的间隔，改了300ms还是7秒左右，是不是我写的有问题？你有这个millis()的测试代码吗？
 谢谢了  又问了你一个问题  麻烦了

FASTSHIFT, 2020-04-03:
@kehale 
millis不对可能是主频不正常导致的，可以进入debug查看SystemCoreClock(主频)的值。
您用的是什么型号的芯片，晶振的频率是多少？

kehale, 2020-04-03:
delay 完美运行 就是millis  我的芯片stm32f030,主频48的。这个 和外部晶振8M没有关系吧，谢谢了。

kehale, 2020-04-03:
[image]
[image]
[image]
我外部用的8M晶振，这个晶振设置没问题，delay准确没问题。还有那个SystemCoreClock（主频），我在debug怎么找不到呢？麻烦给个截图 ，谢谢

FASTSHIFT, 2020-04-03:
[image]
进入debug调试界面，在Watch窗口里输入SystemCoreClock，就可以看到主频了。
delay正常，说明主频是正常的，应该是你写的部分出了问题。
如果实在解决不了的话加我Q吧，个人主页那个邮箱就是。

kehale, 2020-04-03:
好的，能给我一下你写的millis()代码吗？我测试下 谢谢 主要是新手，感觉这个keil arduino非常好，

FASTSHIFT, 2020-04-03:
#define LED_Pin       PC13
#define LED_FlashTime 1000

static uint32_t lastTime = 0;

void setup()
{
    pinMode(LED_Pin, OUTPUT);
}

void loop()
{
    if(millis() - lastTime >= LED_FlashTime)
    {
        lastTime = millis();
        togglePin(LED_Pin);
    }
}

jackliuwenli, 2020-06-12:
你好，大佬，可以建个QQ交流群吗？找不到组织啊~

FASTSHIFT, 2020-06-12:
@jackliuwenli 
私聊我就行，QQ在我的个人主页

closed 2021-10-13 by FASTSHIFT