FASTSHIFT/Arduino-For-Keil #15: [Help wanted] I2C problem

klerone, 2020-10-28:
Hello FASTSHIFT.

I am a hobbyist of Arduino and recently discovered your job and I have been playing with it for 2 or 3 days, I blinked a LED, fade in and out and printed to serial using your examples, so far, so good.

I bought some months ago a board with STM32F030F4P6 (SSOP-20 package), really cheap and I have been working to move some of attiny84 projects to this MCU, not luck with it, code was too big until I found your project.

Now I am trying to Port some of the Arduino libraries for my sensors and screens, many of them are I2C (BMP085, BME280, OLED SSD1306...) So first I tried to use the I2C scanner code from the Arduino community but seems it hangs up the whole thing and don't even get serial Port outputs. 

I will be very happy if you can help me on it.

This is the code:

`//I2C Scanner
//#define pinSDA PA1
//#define pinSCL PA2
#define SDA_Pin PA1
#define SCL_Pin PA2
 
#include <Wire.h>
//TwoWire Wire(pinSCL, pinSDA, SOFT_FAST);
 
void Serial_EventHandler()
{
    togglePin(LED_Pin);
}
 
void setup()
{
  Wire.begin();
  Serial.begin(9600);
                //Serial.setTimeout(10);
  //Serial.attachInterrupt(Serial_EventHandler);
  Serial.println("\nI2C Scanner");
}
 
 
void loop()
{
  byte error, address;
  int nDevices;
 
  Serial.println("Scanning...");
 
  nDevices = 0;
  for(address = 1; address < 127; address++ )
  {
    // The i2c_scanner uses the return value of
    // the Write.endTransmisstion to see if
    // a device did acknowledge to the address.
    Wire.beginTransmission(address);
    error = Wire.endTransmission();
 
    if (error == 0)
    {
      Serial.printf("I2C device found at address 0x");
      if (address<16)
        Serial.printf("0");
      Serial.print(address,HEX);
      Serial.println("  !");
 
      nDevices++;
    }
    else if (error==4)
    {
      Serial.printf("Unknow error at address 0x");
      if (address<16)
        Serial.printf("0");
      Serial.println(address,HEX);
    }   
  }
  if (nDevices == 0)
    Serial.println("No I2C devices found\n");
  else
    Serial.println("done\n");
 
  delay(5000);           // wait 5 seconds for next scan
}
 
int main(void)
{
    Delay_Init();
    setup();
    for(;;)loop();
} `

This is my board:
By the way, the code use the internal oscillator or external? 

[image]

[image]

FASTSHIFT, 2020-10-29:
@klerone 
1. If you want to modify the default software I2C pins, you need to modify the pins in the "Wire.h" file.
2. The 8M external crystal oscillator is used by default. If you want to use the built-in crystal oscillator, you need to call the "InternalClocks_Init()" function before the "Delay_Init()" function.
3. If you want to download programs larger than 16KB into "STM32F030F4P6", you can try to change the MCU selection to "STM32F030C6" in "Options for Target->Device".
[image]

Hope to help you.

klerone, 2020-10-29:
Great! 
Now it's working, I read the BMP085 using adafruit's library, but there is a problem with function pow(), anyway I just cancelled that part because it used too much resources.

You are telling me that 16Kb is not the real flash in this MCU? I loaded one that the compiler said is 18kb, I thought is the compiler was calculating wrong. 

Thank you for your support. 

You are the man!

FASTSHIFT, 2020-10-29:
@klerone 
The FLASH of "STM32F030F4P6" is actually 32KB. The manufacturer only tested the space of the first 16KB, and the stability of the latter 16KB is not guaranteed, but it can be used.

klerone, 2020-11-04:
@FASTSHIFT 

I have problems to get the interrupts working, my board has a button connected to PA0, it has a PULLUP resistor and the button is active LOW.

I'm using the code from the example EXTI, with the changes for my settings.

It just do nothing when i push the button, nothing happens. 
the button is fine, i tested it using the button in the arduino style (reading the status inside the loop).

Can you please point me in the right direction?

`//External interrupts
#include "Arduino.h"

#define LED_Pin PA4
#define KEY_Pin PA0

void LED_Toggle()
{
    //Serial.println("KEY is pressed!");
    togglePin(LED_Pin);
}

void setup()
{
    Serial.begin(57600);
    pinMode(LED_Pin, OUTPUT);
		digitalWrite(LED_Pin, HIGH);
    pinMode(KEY_Pin, INPUT);
    attachInterrupt(KEY_Pin, LED_Toggle, FALLING);
	  //attachInterrupt(digitalPinToInterrupt(KEY_Pin), LED_Toggle, FALLING);
	
}

void loop()
{ }`

FASTSHIFT, 2020-11-04:
@klerone 
EXTI problem has been solved, please update.

klerone, 2020-11-05:
Great, it works!

Thank you!

klerone, 2020-11-06:
@FASTSHIFT 

I have been testing different libraries, all work. but since yesterday I got a problem that i have not idea how to solve.

with the new release of the code suddenly started to get this error:
**FCARM - Output Name not specified, please check 'Options for Target - Utilities'**

I checked if i change something by mistake and all looks fine, then I downloaded one more copy of the example and compile, i works, then i added a library and could not compile.

before this started to happen all was OK, I loaded many libraries and compile and test. 

the old release is fine.

I hope you can point me in the right direction, I cannot find too much help in keil's website.

FASTSHIFT, 2020-11-06:
@klerone 
You can try to recreate the project manually, and then import the necessary files.

klerone, 2020-11-07:
Ok, problem solved.

One more question, is it possible to use the PF0 and PF1 as normal GPIO? I tried to blink a LED, it did not work.

namkihop, 2021-01-23:
[image]
[image]
[image]

Blue pill I2c not found on your library but on Arduino I2C is 0x27.

FASTSHIFT, 2021-01-23:
@namkihop 
Try to reduce the I2C rate.
You can try to comment out the FULL_SPEED_I2C macro definition of Wire.h. If communication still cannot be established, you can try to set SOFT_FAST -> SOFT_STANDARD of the Wire instance at the end of Wire.cpp to further reduce the I2C rate.

namkihop, 2021-01-23:
[image]
tôi đã thử nhưng nó không hoạt động, bạn có mẫu ví dụ nào về i2c không?

FASTSHIFT, 2021-01-23:
@namkihop 
Bạn có thể thử các thiết bị I2C khác không? Đảm bảo hủy FULL_SPEED_I2C.

namkihop, 2021-01-23:
[image]
it worked, let me ask is this i2c soft or hard? thank you.

FASTSHIFT, 2021-01-23:
@namkihop 
Software I2C.