FASTSHIFT/Arduino-For-Keil #16: [user contribution] Libraries - Tested

klerone, 2020-11-17:
@FASTSHIFT 
I want to share the libraries I use and that I tested with the real hardware.

My setup: SMT32F030 with internal clock @8Mhz

I use this libraries, sensors and displays in other arduino projects and I want to share it.
All this libraries are working, I did changes in some of them in order to work correctly, also I added the examples as I use in Keil.
 
List:
_**Sensors**_
**BMP085**                                     [Adafruit_BMP085](https://github.com/FASTSHIFT/Arduino-For-Keil/files/5551770/Adafruit_BMP085_Library.zip)
**BME280**                                    [ Grove_-_Barometer_Sensor_BME280](https://github.com/FASTSHIFT/Arduino-For-Keil/files/5551774/Grove_-_Barometer_Sensor_BME280.zip)
**SHT31**                                        [Adafruit_SHT31](https://github.com/FASTSHIFT/Arduino-For-Keil/files/5551772/Adafruit_SHT31_Library.zip)

_**Display**_                                     
**OLED SSD1306**                       [ Adafruit_SSD1306_extra](https://github.com/FASTSHIFT/Arduino-For-Keil/files/5551773/Adafruit_SSD1306_extra.zip) _(note, this is a mod from the original library to support the 64x48 screen)_ 
                                                & also need the [Adafruit_GFX_Library](https://github.com/FASTSHIFT/Arduino-For-Keil/files/5551771/Adafruit_GFX_Library.zip) (attached here too).
**PCD8544 (Nokia 3310)**            [pcd8544](https://github.com/FASTSHIFT/Arduino-For-Keil/files/5551801/pcd8544.zip) _(not adafruit, this is a lightweight alternative, easy to use)._

_**Wireless**_                                   
**RFM69**                                     [RFM69_LowPowerLab](https://github.com/FASTSHIFT/Arduino-For-Keil/files/5551856/RFM69_LowPowerLab.zip).

_**Others**_                                     
**JSON**                                      [ jWrite](https://github.com/FASTSHIFT/Arduino-For-Keil/files/5551800/jWrite.zip)  _(to make JSON format data_)

[pcd8544.zip](https://github.com/FASTSHIFT/Arduino-For-Keil/files/5551801/pcd8544.zip)
[RFM69_LowPowerLab.zip](https://github.com/FASTSHIFT/Arduino-For-Keil/files/5551856/RFM69_LowPowerLab.zip)
[Adafruit_BMP085_Library.zip](https://github.com/FASTSHIFT/Arduino-For-Keil/files/5551770/Adafruit_BMP085_Library.zip)
[Adafruit_GFX_Library.zip](https://github.com/FASTSHIFT/Arduino-For-Keil/files/5551771/Adafruit_GFX_Library.zip)
[Adafruit_SHT31_Library.zip](https://github.com/FASTSHIFT/Arduino-For-Keil/files/5551772/Adafruit_SHT31_Library.zip)
[Adafruit_SSD1306_extra.zip](https://github.com/FASTSHIFT/Arduino-For-Keil/files/5551773/Adafruit_SSD1306_extra.zip)
[Grove_-_Barometer_Sensor_BME280.zip](https://github.com/FASTSHIFT/Arduino-For-Keil/files/5551774/Grove_-_Barometer_Sensor_BME280.zip)
[jWrite.zip](https://github.com/FASTSHIFT/Arduino-For-Keil/files/5551800/jWrite.zip)

FASTSHIFT, 2020-11-17:
@klerone 
Thanks for your contribution

klerone, 2021-11-04:
LowPower Library for STM32f0xx, easy to use.

 Hope it is useful for someone else.

[LowPower.zip](https://github.com/FASTSHIFT/Arduino-For-Keil/files/7475448/LowPower.zip)

FASTSHIFT, 2021-11-05:
> LowPower Library for STM32f0xx, easy to use.
> 
> Hope it is useful for someone else.
> 
> [LowPower.zip](https://github.com/FASTSHIFT/Arduino-For-Keil/files/7475448/LowPower.zip)

Thanks for your contribution.

closed 2026-04-17 by FASTSHIFT