FASTSHIFT/Arduino-For-Keil #32: Keil5不能引用Arduino库中的Class

lane2022517, 2022-08-05:
*** Using Compiler '', folder: 'D:\Kil_v5\ARM\ARMCC\bin\'
Build target 'Project'
compiling main.cpp...
..\USER\main.cpp(12): error:  #289: no instance of constructor "HardwareSerial::HardwareSerial" matches the argument list
            argument types are: (Pin_Type, Pin_Type)
  HardwareSerial  PB(PA10,PA9);

FASTSHIFT, 2022-08-05:
什么库？Serial对象是提前创建好的，为什么要自己创建？

lane2022517, 2022-08-05:
这是Arduino引用的 HardwareSerial  PB(PA10,PA9)，定义串口1 的接收和发送引脚
但是在Keil5中编译会出现这样的错误 ：
 ..\USER\main.cpp(12): error:  #289: no instance of constructor "HardwareSerial::HardwareSerial" matches the argument list
            argument types are: (Pin_Type, Pin_Type)

lane2022517, 2022-08-05:
> 
这是Arduino引用的 HardwareSerial  PB(PA10,PA9)，定义串口1 的接收和发送引脚
但是在Keil5中编译会出现这样的错误 ：
 ..\USER\main.cpp(12): error:  #289: no instance of constructor "HardwareSerial::HardwareSerial" matches the argument list
            argument types are: (Pin_Type, Pin_Type)

lane2022517, 2022-08-05:
> 什么库？Serial对象是提前创建好的，为什么要自己创建？

这是Arduino引用的 HardwareSerial  PB(PA10,PA9)，定义串口1 的接收和发送引脚
但是在Keil5中编译会出现这样的错误 ：
 ..\USER\main.cpp(12): error:  #289: no instance of constructor "HardwareSerial::HardwareSerial" matches the argument list
            argument types are: (Pin_Type, Pin_Type)
可能是我搞错原因了

closed 2022-08-05 by lane2022517