FASTSHIFT/Arduino-For-Keil #33: 编译无法使用C++11  arduino的很多库在头文件中就给成员赋值了  导致出错   这个问题怎么解决

kaihuaa, 2022-11-29:
编译无法使用C++11  arduino的很多库在头文件中就给成员赋值了  导致出错   这个问题怎么解决

FASTSHIFT, 2022-11-29:
1.对使用C++ 11的cpp文件添加`--cpp11`参数。
2.修改成员变量初始化方式，在构造函数中进行初始化，不在class定义变量的时候直接初始化。

kaihuaa, 2022-11-29:
在对加入的c++文件这里单独配置misc controls  加入--cpp11 文件math.h  报错

FASTSHIFT, 2022-11-29:
用的什么库，可以发一下吗？

kaihuaa, 2022-11-29:
ICM42688

FASTSHIFT, 2022-11-29:
三个修改：
1.给Libraries组添加`--cpp11`参数
[image]
2.在`ICM42688.cpp`中添加宏定义`#define abs(x) ABS(x)`
[image]

3.为该库的引用者（以`main.cpp`为例）添加`--cpp11`参数
[image]

FASTSHIFT, 2022-11-29:
`WMath.h`中引用了`math.h`库，删除即可。还有删除`ICM42688.h`中的`using ICM42688::ICM42688;`
[ICM42688_F103_Test.zip](https://github.com/FASTSHIFT/Arduino-For-Keil/files/10111088/ICM42688_F103_Test.zip)

目前主线维护的是AT32的支持，STM32用得比较少了。

kaihuaa, 2022-11-29:
好的 感谢您 百忙之中 回复我  ，对您的帮助再次表示感谢

closed 2022-12-10 by FASTSHIFT