FASTSHIFT/Arduino-For-Keil #36: 按照教程移植，但是导入库失败

15192397971, 2023-01-18:
教程如下
https://blog.csdn.net/m0_50926303/article/details/120585496
下方为错误信息
Build started: Project: proj
*** Using Compiler 'V5.06 update 6 (build 750)', folder: 'C:\Keil_v5\ARM\ARMCC\Bin'
Build target 'Project'
compiling SPFD5408_Adafruit_GFX.cpp...
..\Libraries\SPFD5408-master\SPFD5408_Adafruit_GFX.cpp(37): error:  #5: cannot open source input file "SPFD5408_Adafruit_GFX.h": No such file or directory
  #include <SPFD5408_Adafruit_GFX.h>
..\Libraries\SPFD5408-master\SPFD5408_Adafruit_GFX.cpp: 0 warnings, 1 error
compiling SPFD5408_TouchScreen.cpp...
..\Libraries\SPFD5408-master\SPFD5408_TouchScreen.cpp(6): error:  #5: cannot open source input file "pins_arduino.h": No such file or directory
  #include "pins_arduino.h"
..\Libraries\SPFD5408-master\SPFD5408_TouchScreen.cpp: 0 warnings, 1 error
compiling SPFD5408_Adafruit_TFTLCD.cpp...
..\Libraries\SPFD5408-master\SPFD5408_Adafruit_TFTLCD.h(18): error:  #5: cannot open source input file "SPFD5408_Adafruit_GFX.h": No such file or directory
  #include <SPFD5408_Adafruit_GFX.h>
..\Libraries\SPFD5408-master\SPFD5408_Adafruit_TFTLCD.cpp: 0 warnings, 1 error
".\Objects\proj.axf" - 3 Error(s), 0 Warning(s).
Target not created.

FASTSHIFT, 2023-01-18:
硬件是什么平台？
是基于这个库吗：https://github.com/JoaoLopesF/SPFD5408

15192397971, 2023-01-18:
原本的硬件是arduino uno 打算把程序移植到stm32f103c8 是基于这个库https://www.yiboard.com/forum.php?mod=viewthread&amp;tid=763&amp;highlight=%E8%A7%A6%E6%91%B8%E5%B1%8F%E8%AE%A1%E7%AE%97%E5%99%A8程序类似于这个
 

悲催的骚年
***@***.***

FASTSHIFT, 2023-01-18:
原始库和底层寄存器耦合太紧密，不方便移植。
我另外找了一个版本重新移植：https://github.com/nopnop2002/STM32_TFT_8bit 请参考这个链接的说明进行连线。
由于没有硬件无法进行测试，所以不保证可以正常工作。
Keil工程：[SPFD5408_F103.zip](https://github.com/FASTSHIFT/Arduino-For-Keil/files/10446727/SPFD5408_F103.zip)

15192397971, 2023-01-18:
好的谢谢大佬，我明早起来试一试 

悲催的骚年
***@***.***